BCN3D/Stratos #5: Segfault on start on linux

nallath, 2021-07-14:
**Application version**
1.0.0 

**Platform**
Fedora 32

**Printer**
Didn't get that far. 

**Reproduction steps**
Try to start the application image.

**Actual results**
Nothing happens, segfault. 

**Expected results**
Application starts. 

**Log file**

Fatal Python error: Segmentation fault

Thread 0x00007f4e90a79700 (most recent call first):
  File "/home/ubuntu/cura-install/lib/python3.8/threading.py", line 302 in wait
  File "/home/ubuntu/cura-install/lib/python3.8/threading.py", line 433 in acquire
  File "/home/ubuntu/stratos-build-intern/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/home/ubuntu/stratos-build-intern/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 123 in run
  File "/home/ubuntu/cura-install/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/home/ubuntu/cura-install/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f4e9127a700 (most recent call first):
  File "/home/ubuntu/cura-install/lib/python3.8/threading.py", line 302 in wait
  File "/home/ubuntu/cura-install/lib/python3.8/threading.py", line 433 in acquire
  File "/home/ubuntu/stratos-build-intern/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/home/ubuntu/stratos-build-intern/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 123 in run
  File "/home/ubuntu/cura-install/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/home/ubuntu/cura-install/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f4e91a7b700 (most recent call first):
  File "/home/ubuntu/cura-install/lib/python3.8/threading.py", line 302 in wait
  File "/home/ubuntu/cura-install/lib/python3.8/threading.py", line 433 in acquire
  File "/home/ubuntu/stratos-build-intern/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/home/ubuntu/stratos-build-intern/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 123 in run
  File "/home/ubuntu/cura-install/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/home/ubuntu/cura-install/lib/python3.8/threading.py", line 890 in _bootstrap

Thread 0x00007f4e9227c700 (most recent call first):
  File "/home/ubuntu/cura-install/lib/python3.8/threading.py", line 302 in wait
  File "/home/ubuntu/cura-install/lib/python3.8/threading.py", line 433 in acquire
  File "/home/ubuntu/stratos-build-intern/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/home/ubuntu/stratos-build-intern/build/inst/lib/python3/dist-packages/UM/JobQueue.py", line 123 in run
  File "/home/ubuntu/cura-install/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/home/ubuntu/cura-install/lib/python3.8/threading.py", line 890 in _bootstrap

Current thread 0x00007f4ef67a6740 (most recent call first):
  File "/home/ubuntu/stratos-build-intern/build/inst/lib/python3/dist-packages/cura/UI/CuraSplashScreen.py", line 73 in drawContents
  File "/home/ubuntu/stratos-build-intern/build/inst/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 546 in showSplashMessage
  File "/home/ubuntu/stratos-build-intern/build/inst/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 203 in startSplashWindowPhase
  File "/home/ubuntu/stratos-build-intern/build/inst/lib/python3/dist-packages/cura/CuraApplication.py", line 473 in startSplashWindowPhase
  File "/home/ubuntu/stratos-build-intern/build/inst/lib/python3/dist-packages/UM/Application.py", line 239 in run
  File "/home/ubuntu/stratos-build-intern/build/inst/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 341 in run
  File "/home/ubuntu/stratos-build-intern/build/inst/lib/python3/dist-packages/cura/CuraApplication.py", line 779 in run
  File "/home/ubuntu/stratos-build-intern/build/inst/bin/BCN3D_Stratos", line 235 in <module>
  File "/home/ubuntu/cura-install/lib/python3.8/site-packages/cx_Freeze/initscripts/Console.py", line 36 in run
  File "/home/ubuntu/cura-install/lib/python3.8/site-packages/cx_Freeze/initscripts/__startup__.py", line 74 in run

**Additional information**
The template refers to the information in the Cura repository, which is not where the logging for Stratos is located.

Ghostkeeper, 2021-07-14:
Strange, I'm getting a different error:

```
Traceback (most recent call last):
  File "/home/ubuntu/cura-install/lib/python3.8/site-packages/cx_Freeze/initscripts/__startup__.py", line 74, in run
  File "/home/ubuntu/cura-install/lib/python3.8/site-packages/cx_Freeze/initscripts/Console.py", line 36, in run
  File "/home/ubuntu/stratos-build-intern/build/inst/bin/BCN3D_Stratos", line 25, in <module>
ImportError: libnlopt.so.0: cannot open shared object file: No such file or directory
```

marcmateo, 2021-07-20:
Hello @nallath and @Ghostkeeper ,

We are already working on the next release and we look forward to solving these problems. When you say that the template goes to the Cura repository and not Stratos, where exactly do you mean ?

Regards,
Marc

nallath, 2021-07-20:
Oh, the template links to a cura link on "how to get the log files". But those instructions don't work for stratos, since the files are located somewhere different.

marcmateo, 2021-10-18:
Hello @nallath and @Ghostkeeper 

I already fixed the "how to get the log files" issue and the compatibility problems with Ubuntu. I hope that you no longer have these problems with the last release.

Regards,
Marc