BCN3D/Stratos #8: Preheat issue, PVA cloggs, BVOH compatibility

Pyrgator, 2021-07-20:
When printing using both hotends, one for main material other for PVA/BVOH support the printer preheats both hotends to printing temperature instead of one to standby and other to printing. With big prints this causes PVA to clogg the nozzle.
Also during the print the printer never cools to standby temperature, just cools down a little, also causing PVA to clogg the nozzle.
BVOH isnt supported in Stratos. Both PVA and BVOH profiles need tuning, for example fan speed is too high for both filaments.

**Application version**
1.0

**Platform**
Win10

**Printer**
Sigmax R19

LuisCRSousa, 2021-07-20:
I'm having exactly the same problem.
Seems impossible to print with BVOH or PVA. The support extrudder starts heating like magic when's in standby for longs periods of time and it gets clogged right away after.

Pyrgator, 2021-07-23:
Are you guys going to reply?

mpikmwm, 2021-07-27:
got the same problem, right extruder clogged, after printig with bvoh

Pyrgator, 2021-07-27:
Hey, are you serious? Is this company real? Where is any answer?

marcmateo, 2021-07-27:
Hello,

Thanks for the report, we are working to solve the problems you mentioned to be able to integrate it in the next release, for now, we recommend you to lower the "Standby Temperature" setting from 150ºC to 100ºC, with this change you should notice an important improvement. If the problem persists after changing this, please contact us and we will try to solve it as soon as possible.
 
Regards,
Marc

Pyrgator, 2021-07-27:
Lowering the temperature doesnt work, I have already tried that. To integrate it into slicer-  you promised this when there was "old" BCN-Cura. The problem persists, this is how Im contacting you.

marcmateo, 2021-07-27:
Hello @Pyrgator ,

We also recommend adjusting the Start Purge Speed setting up to 5mm/s on both heads. If you still cannot get the desired results or any other question, do not hesitate to contact us.

Regards,
Marc

marcmateo, 2021-07-29:
Hello @Pyrgator ,

If you are still having printing problems, please send us the Stratos Project File so we can analyze it and find a better solution. In the following [link ](https://www.bcn3d.com/bcn3d-support/)you will find the form to contact us and send the project, as well as other information to help you better.

Regards,
Marc

closed 2021-09-15 by marcmateo