BCN3D/Stratos #9: Unable to start Stratos in Ubuntu 18.04

x800, 2021-07-29:
**Application version**
BCN3D_Stratos-1.0.0.AppImage

**Platform**
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic

**Printer**
SigmaX

**Reproduction steps**
./BCN3D_Stratos-1.0.0.AppImage 
Traceback (most recent call last):
  File "/home/ubuntu/cura-install/lib/python3.8/site-packages/cx_Freeze/initscripts/__startup__.py", line 74, in run
  File "/home/ubuntu/cura-install/lib/python3.8/site-packages/cx_Freeze/initscripts/Console.py", line 36, in run
  File "/home/ubuntu/stratos-build-intern/build/inst/bin/BCN3D_Stratos", line 25, in <module>
ImportError: /tmp/.mount_BCN3D_36ivSM/usr/bin/lib/pynest2d.so: undefined symbol: nlopt_get_errmsg

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/BCN3D/Stratos/wiki/Logging-Issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

marcmateo, 2021-07-30:
Hello,

We are currently working with the problems found in certain versions of Linux, we hope to solve it as quickly as possible.

Regards,
Marc

mpikmwm, 2021-09-13:
With the new version, still not able to start on Ubuntu!
`Traceback (most recent call last):
  File "/home/ubuntu/cura-install/lib/python3.8/site-packages/cx_Freeze/initscripts/__startup__.py", line 74, in run
  File "/home/ubuntu/cura-install/lib/python3.8/site-packages/cx_Freeze/initscripts/Console.py", line 36, in run
  File "/home/ubuntu/stratos-build-intern/projects/inst/bin/BCN3D_Stratos", line 25, in <module>
ImportError: libnlopt.so.0: cannot open shared object file: No such file or directory
`

marcmateo, 2021-09-14:
Hello,

I just updated the Ubuntu version and you should no longer have problems to start Stratos in Ubuntu . If the problem persists, please let us know.

Regards,
Marc

closed 2021-09-27 by marcmateo