BCN3D/Stratos #10: Stratos very weird behaviour on print BCN3D Sigma

WanaGo, 2021-07-30:
**Application version**
Stratos 1.0.0

**Platform**
Windows 10 PRO

**Printer**
BCN3D Sigma (2016/2017 version)

**Reproduction steps**
Left extruder only, printed a part I designed in Solidworks, exported as STL, ABS, Support. Printing from SD card.

**Screenshot(s)**
[image]
[image]

**Actual results**
Installed Stratos yesterday, and did a print this morning, and all went quite well. Did another print this evening, and half way through it started going the full Y direction, all the way to the front of the bed, when the print was at the very back, and also paused itself. I resumed it, thinking wtf just happened, and then it did it again, went all the way to the front of the bed, making big long strings, and then went all the way to the left and just stopped. moving, print still running but doing nothing. I stopped the print.
All those strings are errors, they should not have happened.
It should not have paused.
It should not have stopped.
I have no idea what just happened.

**Expected results**
The print should have completed, not paused half way, not gone all the way to the front of the bed causing the big strings in the printer, etc. Very weird.

**Project file**
Attached .mf file
[Shroud-Noctua-A6x25.zip](https://github.com/BCN3D/Stratos/files/6907485/Shroud-Noctua-A6x25.zip)

**Log file**
N/A

**Additional information**
N/A

starstorms9, 2021-08-02:
I don't have a nicely documented case like @WanaGo but I have definitely seen similar things happen on smaller parts with my Sigma R19 and Stratos 1.0.0, but it's quite rare. It would just all of a sudden go a bit extra in one direction and then resume printing over air. At first I thought it was a mechanical / motor issue and it skipped a layer or something but having watched it happen a few times it definitely appears deliberate and is repeatable with the same gcode file. And it isn't a sudden skip, it's a smooth motion to the new (wrong) location as if it were part of the program.

marcmateo, 2021-08-02:
Hello,

Do you have the gcode that you have generated to test it in our laboratory ? I made a gcode from your project and it does not make me those movements, also to be able to slice it, I had to change the build plate temperature, have you touched any more settings before genetaring the gcode ?

Regards,
Marc

WanaGo, 2021-08-02:
Hello,
Yes I set the build plate max to 115C, had to increase the max by simply editing the bcn3dsigma.def.json file
"material_bed_temperature": { "maximum_value": 115, "maximum_value_warning": 110 },
I use a Filafarm alunimum bed and Filaprint top, and it needs to be around 115 for ABS to have really good success for the print. I have an upgraded bed heating system so I can reach that temperature OK. Not related to this problem though.

Nothing else has been changed, other than the settings in the profile.

I will check my SD card in the morning to see if I still have the gcode that did this.

I have not been able to replicate the issue after doing a few prints today and yesterday in Stratos. Very odd indeed what happened. Never before encountered that on previous versions of Cura.

Regards

WanaGo, 2021-08-05:
Sadly the file was overwritten, and I have not been able to reproduce this again yet.

closed 2021-09-10 by marcmateo