BCN3D/Stratos #11: Dual extruder mode initial layers have extrusion amount incorrect

starstorms9, 2021-08-02:
**Application version**
BCN3D Stratos 1.0.0

**Platform**
Windows 10

**Printer**
Sigma R19

**Reproduction steps**
Slice the model in dual extrusion mode and save the gcode file.

**Screenshot(s)**
[image]

**Actual results**
It outputs the gcode successfully but when printing the file the extrusion amounts are incorrect. This can be seen in the gcode as show below.

**Expected results**
The first layers should start printing with extrusion at 8 mm ("the nozzle switch retraction distance") but instead they start at E0 which makes them underextrude.

**Project file**
[project_file_and_gcode.zip](https://github.com/BCN3D/Stratos/files/6914335/project_file_and_gcode.zip)

**Log file**
[BCN3D Stratos.log](https://github.com/BCN3D/Stratos/files/6914343/BCN3D.Stratos.log)

**Additional information**
I started using Stratos 1.0.0 recently and ran into a problem with dual extrusion mode on my Sigma R19 where it would always start the first layer not extruding enough initially and to get it to work I would have to manually pause and purge the extruders out every print on the first layer so they would be properly primed in again. It would literally go from retracting over the purge bin to directly start printing with no priming. After that though, all subsequent layers work fine.

To narrow this issue down, I re-downloaded Stratos, started from all stock settings, etc and sliced an extremely simple dual extrusion model that is just several 0.2mm tall rectangles layered on top of eachother with each layer assigned to alternating extruders and sliced at 0.2mm layer height so that when it prints it just alternates between the two extruders every layers and does a simple small rectangle, no skirt / brim / infill / extra walls / top bottom layers / etc. I attached the output gcode for this in the attached 'layers_stock.gcode' inside the project zip file.

I figured out that the issue is fairly fundamental and I'm quite confused what the intent is supposed to be here or if this is indeed a bug. Basically, for the first time each extruder is used it starts printing with the extrusion amount starting from exactly 0mm. See excerpt from the attached gcode file below. When it starts printing WALL-OUTER it's exactly 8mm deficient on filament. If I comment out the G92 E0 line then it all works fine.

**T1**
G92 E0
G91
G1 Z0.2 F12000
G90
G92 E-8
G1 E0 F1000 ;Restore E position
G1 E0.8 F100;defaultpurge
G92 E0
G4 S3
**G1 E-8 F1000
G92 E0**
M105
M109 S190
M104 T0 S192.5
M106 S51
M104 T0 S195
M204 S690
M205 X10.94 Y10.94
;MESH:dual_layered_200um_top.STL
G0 F5850 X105 Y297 Z0.8
G1 F600 Z0.4
G0 F5850 X105 Y147.78
G0 X109.7 Y147.73
G0 X109.8 Y147.8
M204 S352
M205 X7.03 Y7.03
;TYPE:WALL-OUTER
**G1 F1050 E0**
G1 F900 X100.2 Y147.8 **E0.12039**
G1 X100.2 Y143.2 E0.17807
G1 X109.8 Y143.2 E0.29846
G1 X109.8 Y147.8 E0.35615

So what it should do is start at E8 when it actually starts printing (which is the nozzle switch retraction distance) instead of 0 because it had just retracted to do the nozzle switch retract and then did a G92 E0 so going to 0 will result in no extrusion. The rest of the layers do this properly as shown below:

T1
G92 E0
G91
G1 Z0.2 F12000
G90
G92 E-8
G1 E0 F1000 ;Restore E position
G1 E0.8 F100;defaultpurge
G92 E0
G4 S3
**G1 E-8 F1000
G92 E0**
M106 S153
;MESH:dual_layered_200um_top.STL
G0 F9000 X109.8 Y147.8 Z1.2
M204 S270
M205 X4.69 Y4.69
;TYPE:WALL-OUTER
G1 F600 Z0.8
**G1 F525 E8**
G1 F900 X100.2 Y147.8 E8.12039
G1 X100.2 Y143.2 E8.17807
G1 X109.8 Y143.2 E8.29846
G1 X109.8 Y147.8 E8.35615

I can't simply edit the extruder start gcode because it's the actual printing extrusion code that is off so there's not much I can do. I can't imagine that this is a bug that just straight up prevents doing dual extrusion prints properly but I'm also at a loss why it would be like this, does anyone have an idea and how to fix this properly?

Thanks!

marcmateo, 2021-08-03:
Hello @starstorms9 ,

While we look to solve the problem for the next release, we recommend that you use the skirt or the brim so that these first layers where the error happens do not influence the final result.

Regards,
Marc

starstorms9, 2021-08-08:
Thanks for the update @marcmateo! I actually found out that enabling 'relative extrusion' in the special modes category fixes this problem so I've been using that.

marcmateo, 2021-08-10:
Hello @starstorms9 ,

We are glad to know that you have already been able to solve it, for any other questions or problems do not hesitate to contact us.

Regards,
Marc

closed 2021-08-10 by marcmateo