BCN3D/Stratos #14: Unable to slice

Pyrgator, 2021-11-02:
**Application version**
1.2.2

**Platform**
win10

**Printer**
Sigmax r19

**Reproduction steps**
Tried to slice... wasnt able.

Unable to slice

Please review settings and check your models.
- Fit within the building volume- Fits
- Are assigned to an enabled extruder- are assigned
- Are not set as modifier meshes- no they arent.

If I use only one extruder for both support and main print, slicer slices.

**Screenshot(s)**

[image]

**Actual results**

Slicer doesnt slice

**Expected results**

Slice that damn thing

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/BCN3D/Stratos/wiki/Logging-Issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

marcmateo, 2021-11-03:
Hello @Pyrgator ,

Could you detail all the steps you have followed to get to this error ? If you prefer, you can send the project [here ](https://support.bcn3d.com/knowledge/kb-tickets/new) so that we can try to find a solution more adapted to your problem.

For any other question, do not hesitate to contact us.

Regards,
Marc

Pyrgator, 2021-11-22:
Hi Marc,

I loaded the file and tried to slice, nothing special, printmode- one extruder ABS and another BVOH, when I wiggled the model the problem went away, I literally moved it to another place and then back and it worked. When I experience this again I will let you know.

Have a nice day