BCN3D/Stratos #15: Both extruders preheat... STILL

Pyrgator, 2021-11-02:
**Application version**
1.2.2

**Platform**
win 10

**Printer**
Sigmax r19

**Reproduction steps**
Both extruders preheat to print temperature at the begining of print instead of one staying in standby temperature. This issue still presists in despite of you promising to fix it in the last version. Im tired of manually modifying gcode on 4k machine. Its time to fix it.
Also dont close issure without resolving them

**Screenshot(s)**
(Image showing the problem, perhaps before/after images.) 

**Actual results**
(What happens after the above steps have been followed.)

**Expected results**
(What should happen after the above steps have been followed.)

**Project file**
(For slicing bugs, provide a project which clearly shows the bug, by going to File->Save. For big files you may need to use WeTransfer or similar file sharing sites.)

**Log file**
(See https://github.com/BCN3D/Stratos/wiki/Logging-Issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

marcmateo, 2021-11-03:
Hello @Pyrgator ,

In order to help you solve the problem, I would need you to give me a little more detail about the settings you are using in Stratos, such as the printing mode. 

I recommend that if you are only going to print with an extruder you have to select the Single printing mode (T1 or T2), if you use the Dual printing mode if you only want to print with an extruder you may not obtain the expected results, since this mode is designed to use the both extruders.

For any other question, do not hesitate to contact us.

Regards,
Marc

S3D-r, 2021-11-03:
If you have raised this issue in Github BCN3D Cura, linking that issue also here might be useful.

Pyrgator, 2021-11-22:
Hi Marc,

What details specific details you need? Its quite simple, I want to print an object that needs supports, T0 is ABS T1 is BVOH, at the very beginning of the print both of those hotends heat up to print temp, yes there is STANDBY TEMPERATURE but its ignored. So every time slice something in Stratos I have to open the Gcode and manually change temperature in lines 20 and 23 to standby temperature otherwise the T1 will burn the BVOH in nozzle. T0 is always the first one to start printing. When its small object there isnt a problem but my first layer sometimes takes 20 minutes to lay down and that is way too long for BVOH to withstand. I hope I made myself clear, please stop promissing that in NEXT version it will be fixed, because I heard it already many times. If you need any more details from me to actually fix it, let me know.

Have a nice day

marcmateo, 2021-11-23:
Hello @Pyrgator ,

What type of build plate adhesion are you using ? We have found a problem with the standby temperature with the Skirt setting, if it is the one you are using I recommend that you change it to Raft while we look to solve the problem. We hope we can have it fixed soon, and sorry for the inconvenience, we will let you know when we have it solved.

Regards,
Marc

Pyrgator, 2021-11-24:
Im using skirt and sometimes brim. How is Raft going to help? I mean it even prolongs the time T0 is going to be printing the first layer... Im going to keep changing the gcode and you please do have a look.

Have a nice day

hannes3011, 2021-12-27:
please add sigma r19 printer back to original cura. stratos is a single nightmare,

no material updates, no prifile updates.. etc.. please

its a 6 k maschine not a 100 bug crap china part. 

stratos is the reason we dont upgrade to d25 or epsilon. !!!

Sassw, 2022-11-15:
Is this bug fixed on W50?

We also do not recommend statos to anyone.