BCN3D/Stratos #16: Wrongly calculated "doesn't fit into the build volume" if using brim.

S3D-r, 2021-11-03:
Stratos 1.2.2

I have an item whose max dimensions (X/Y) are a bit less than the size of the build table, but the **footprint is much smaller than the table**. I.e. the object widens as the Z gets bigger. So, there would be plenty of space for a brim. However, if I add even a small brim, Cura immediately claims the item doesn't fit.

This same thing was flagged as "a bug" in BCN3D Cura Github ( https://github.com/BCN3D/Cura/issues/115 ) **more than 3 years ago**. It was never fixed in Cura, and not fixed in Stratos.

marcmateo, 2021-11-03:
Hello @S3D-r ,

Thanks for the report, we will look at it and if there is any kind of error we will put it in the next release.

For any other question, do not hesitate to contact us.

Regards,
Marc

marcmateo, 2021-11-04:
Hellor @S3D-r ,

I have been testing with the maximum dimensions of the bed with objects that occupy the entirety of its space and I do not see any problem. In this image you can see  an object using all the space without using the Brim.

[image]

If you want to use Brim, it is true that the space is reduced, but this space is reduced depending on the settings of the Brim that you want to use, for example, it is possible to put a Brim of 10mm with an object that occupies 410mm in X.

[image]

If you are still having this problem, please send us the Stratos Project File so we can analyze it and find a better solution. In the following [link ](https://support.bcn3d.com/knowledge/kb-tickets/new)you will find the form to contact us and send the project, as well as other information to help you better.

Regards,
Marc

S3D-r, 2021-11-04:
You missed the point. Your object doesn't fit in the bug description. It doesn't widen with Z, which was clearly the key point of the problem. Make a 420mm wide object with 350mm wide footprint and try to have a 10mm wide bring. Doesn't succeed. 

About your example: You have a 410 mm wide footprint with an 8mm wide brim and a 1mm separation between item and print. That will occupy 410 + 2x(8+1) = 428 mm in X, how is that possible if the buildplate is only 420 mm wide?  I tried this, didn't succeed.

marcmateo, 2021-11-04:
Hello @S3D-r ,

I'm not sure what you mean by "420mm wide object with 350mm wide footprint and try to have a 10mm wide bring", do you mean a 420 X 350 Y and 10 Z ?  Are you using the Uniform scaling ?
 
In the example that I have put I have an object with a measure in X of 410mm, I set a Brim Line Count of 10 (this setting overrides the Brim Width, setting the width by the number of printed lines)  that means that the distance that the Brim will occupy in the buildplate is 10 * 0.4 (the size of the hotend, in total 4mm), to this one you must add the value of 1mm that I have set in Brim Distance, in total 5mm per side in X, so my maximum measurement in X is 410mm,  420 = 410 + 2*(10*0.4+1).

Wich printer are you using ?

Regards,
Marc

S3D-r, 2021-11-04:
He's an item that is 420mm wide (X) but its footprint is 264mm (X). You should be able to put an extremely wide brim (say 50mm)... to it, but Stratos accepts none.
[image]

marcmateo, 2021-11-05:
Hello @S3D-r ,

In Stratos the calculation of the bed margin for the Brim is always calculated by with the total measurements of the model in the X / Y axes  and for most of the jobs it has always worked well this way. Anyway we will study it to implement future releases.

Regards,
Marc

S3D-r, 2021-11-05:
I see. However, as mentioned in the opening post, this was flagged (by BCN3D) as "a bug" **3 years ago**...  And honestly - it is a bug.

closed 2021-11-13 by marcmateo

S3D-r, 2021-11-13:
So, what's your opinion, is it a bug or not?

marcmateo, 2021-11-25:
Hello @S3D-r ,

I do not think it is a bug, we have been studying it and for the vast majority of cases it works well as we have it implemented, in your particular case we think that it would be better to turn the object or put supports for greater stability of the impression.

We do not think that implementing brim in the cases you mention it is the solution and it can have also a complex development since many factors would have to be taken into consideration such as the space occupied by the brim and the height of the object or if supports are used they could have problems with the brim itself.

We do not believe that it is the solution to cases like the one you raised, since there are other more friendly solutions like the ones I mentioned at the beginning.

For any other question, do not hesitate to contact us.

Regards,
Marc

S3D-r, 2021-11-25:
Are you kidding me?  In my example, when Stratos claims "_the brim will not fit onto the bed_", the claim is clearly **a false claim**. Do you agree?   The example I showed was just an example, in real-life cases, the part could be more complicated and the suggestion you offered would not apply. And even if they would apply, a bug is a bug even if the user could find a good workaround for it.

So,
- the bug impacts rare cases only
- in most cases, there is a good workaround available
- the bug is difficult to fix without causing side effects
Yet these are _not_ valid reasons to not call it a bug. 

Finally, the new label "wontfix" is OK, for the reasons listed above. Not all bugs need to be fixed if the fixing isn't reasonable.