BCN3D/Stratos #17: Stratos "Report a  Bug" won't lead to this site

S3D-r, 2021-11-03:
In Cura, if you clicked _Help - Report a Bug_, you were sent to the Cura Issues site in Github. But in Stratos, you will be sent to BCN3D homepage, printer issues page. I guess it's not supposed to be like that.

marcmateo, 2021-11-03:
Hello @S3D-r ,

It is not a mistake, we believe that the current contact form in the BCN3D homepage is better for users who are not familiar with the Github platform. 

For any other question, do not hesitate to contact us.

Regards,
Marc

S3D-r, 2021-11-03:
Please accept my strong disagreement. 
1. The form is clearly designed for hardware problems: It asks for a printer serial number and plenty of other information, which are mostly irrelevant when talking about Stratos issues.
2. In the BCN3D site the problems and their solutions seem to be between the issuer and BCN3D only, not public. At least at a glance (don't know that database deeply). Compared to that, the structure and clearness of the Github is superior. Also, Github allows other _users_ to comment on the issue, too, not only the _BCN3D staff_. That gives in many cases faster answer and saves BCN3D effort (when they don't need to solve each problem) when it's rather a user error than a software bug in question.

marcmateo, 2021-11-05:
Hello @S3D-r ,

I understand your reasons, we will rethink it to assess your proposal, thank you for the contribution.

Regards,
Marc

closed 2021-11-10 by marcmateo