BCN3D/Stratos #18: No Print Sequence option in Duplication or Mirror mode

S3D-r, 2021-11-11:
If you choose anything but "Single 1 or 2", you're no longer able to choose Print Sequence but "All at once" will be the only option.  From a technical point, there's no reason to have this limit. Very annoying.

marcmateo, 2021-11-13:
Hello @S3D-r ,

At the moment we do not have it enabled in the modes you comment for security reasons. When using this parameter in mirror / duplication you can have printing problems if the heads are not well calibrated or there is a slight difference in their height or with certain routines of movement of the heads, which could end up having to use layers compensation and recalibrate internal routines of the machine.

Regards,
Marc

S3D-r, 2021-11-13:
Why would miscalibration cause problems in "one at a time" mode but not in "all at once" mode???  Symptoms would be the same in both cases...  I still can't see more severe security issues in "one at a time" mode than in "all at once" mode. 

Printing a compensation layer is also exactly the same process in both of these modes.