BCN3D/Stratos #21: [1.2.2] Appimage Segmentation fault  at usr/bin//stratos.sh: line 14:  8335

eballetbo, 2021-11-26:
**Application version**
1.2.2

**Platform**
Linux (Fedora 35)

**Printer**
None yet

**Reproduction steps**
1. Execute BCN3D_Stratos-1.2.2.AppImage

**Screenshot(s)**
 There is no GUI to take screenshots of.

**Actual results**
 Nothing appears to happen. The process is killed immediately.

When running the AppImage in a terminal, the following error message is shown:
```
<frozen zipimport>:259: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Qt: Session management error: None of the authentication protocols specified are supported
Attribute Qt::AA_UseDesktopOpenGL must be set before QCoreApplication is created.
Fontconfig warning: "/etc/fonts/fonts.conf", line 5: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-hinting-slight.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/11-lcdfilter-default.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/20-unhint-small-vera.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/25-unhint-nonlatin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/30-metric-aliases.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/40-nonlatin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-generic.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/45-latin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/49-sansserif.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/50-user.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/51-local.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-generic.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/60-latin.conf", line 4: unknown element "description"
Fontconfig warning: "/etc/fonts/conf.d/65-nonlatin.conf", line 4: unknown element "description"
Fontconfig warning: FcPattern object weight does not accept value [0 205)
/tmp/.mount_BCN3D_pDqdll/usr/bin//stratos.sh: line 14:  8335 Segmentation fault      (core dumped) BCN3D_Stratos "$@"
```

**Expected results**
 Cura starts.

**Project file**
None

**Log file**
(See https://github.com/BCN3D/Stratos/wiki/Logging-Issues to find the log file to upload, or copy a relevant snippet from it.)

**Additional information**
(Extra information relevant to the issue.)

marcmateo, 2021-11-30:
Hello @eballetbo ,

We are currently working with the problems found in certain versions of Linux, we hope to solve it as quickly as possible.

Regards,
Marc

eballetbo, 2021-12-22:
Can you share build instructions or point me to some documentation on how to build so I can try to run on my machine?

eballetbo, 2022-03-02:
Any news on that? it's being a while since this problem, and seems to affect several Linux distributions if not all.

mpararols, 2022-03-02:
Same problem here with Manjaro

eballetbo, 2022-03-05:
Latest version, 1.4.0 seems to be also affected for this bug.

PhictionalOne, 2023-04-14:
Did you find a workaround? My Sigma is currently just a fancy piece of metal to me...